rapyuta-robotics/RapyutaSimulationPlugins #187: Wrong point cloud data generation from 3D Lidar component

james-yoo, 2023-07-15:
Hi Rapyuta-robotics team,
I am testing an example 3D lidar component(Ouster OS032) with default settings as below:
[image]  
  
When starting the simulation, rays show the correct point clouds as configuration.
[image]  
  
However, rviz shows weird point clouds as below:
[image]

michaelaeriksen, 2023-07-20:
I'm relative new to Unreal and this plugin would you mind sharing how you add the 3d lidar to a scene?

Hisham178, 2023-07-27:
I'm having the same issues. Please let me know if there any work around this

yuokamoto, 2023-07-27:
Sorry for inconvinence
I'm investigating the issue now. Not sure how much time I need to fix since I only have time to investigate in weekend.

yuokamoto, 2023-07-27:
I found cause luckily and created PR.
https://github.com/rapyuta-robotics/RapyutaSimulationPlugins/pull/192

closed 2023-07-31 by yuokamoto