rapyuta-robotics/RapyutaSimulationPlugins #189: Question: How do I import a urdf model?

michaelaeriksen, 2023-07-20:
I'm trying to use your plugin, but I can't seem to find any examples or any documentation for how to use the urdf import capability to import a urdf model into Unreal.  

Any help would be appreciated.

yuokamoto, 2023-07-20:
Thanks for your question.
Currently we are not supporting spawning robot from URDF yet. we have a plan to support it and working on it.

Ckerrr, 2023-07-28:
Hi @yuokamoto , thank you all for the great work to build this plugin. 

About the URDF importer, I saw you have implemented URDF parser in the codebase. Can we know what are the remaining work and an estimated timeline to have the importer work? Thanks ahead

yuokamoto, 2023-07-29:
Currently we are working on creating skeletal mesh from parsed URDF info.
Targeting to complete MVP in this year but not 100% confidence.
cc @Tadinu

james-yoo, 2023-12-21:
@yuokamoto 
Any progress on this feature?

yuokamoto, 2023-12-21:
Unfortunately not major progress since key contributer left this project...
I may make current very beginning prototype public later...

mtmal, 2025-03-06:
Hi @yuokamoto ,

First, apologies for posting under this issue where the last response was 2 years ago. What led me to do this is your last post here:

> Unfortunately not major progress since key contributer left this project...

Could you confirm if Rapyuta Robotics is still maintaining your Unreal Engine 5 - ROS2 integration? 

And to stay a bit on topic - there is a URoboSim plugin for Unreal Engine 4 which supports importing URDF and SDF files: https://github.com/urobosim/URoboSim it might be worth looking into it.

Thanks,
Mateusz

yuokamoto, 2025-03-08:
Thanks for your comment and sorry for not updating info
It is not actively maintened but Rapyuta keep using simulation in internal usage

I'm working on URDF importer and some test video is here.
https://www.youtube.com/watch?v=NFqqZtMhRyc
progress is slow and need to handle more and more cases but this is current status.