rapyuta-robotics/RapyutaSimulationPlugins #193: What distance from the roof to the floor of the OccupancyGridMap is needed for the pgn to be generated correctly?

YuminosukeSato, 2023-07-29:
[image]
What distance from the roof to the floor of the OccupancyGridMap is needed for the pgn to be generated correctly?
I created the roof in Wall_400x300 and everything is black as shown above. So how far away would the roof not be affected?

yuokamoto, 2023-07-29:
you can change MaxVerticalHeight 
https://rapyutasimulationplugins.readthedocs.io/en/latest/doxygen_generated/html/d2/dde/class_a_occupancy_map_generator.html#a4984d0bdf4e3c5347a20f3dbf1c03522

yuokamoto, 2023-08-12:
closed since no activity

closed 2023-08-12 by yuokamoto