rapyuta-robotics/RapyutaSimulationPlugins #194: Frame rate drop

james-yoo, 2023-08-05:
Hi @yuokamoto,
I am testing a sensor component and figuring out that there is a frame drop issue. As you can see in the following video, the frame rate of each sensor drops by about 40% from the set value. 
- ROS2CameraComponent : 15 Hz -> 8~9Hz
- ROS23DLidarComponent : 10 Hz -> 6Hz

[image]

Do you know if there is anything I need to set up to address this?

yuokamoto, 2023-08-05:
How about the /clock topic rate or real time factor of the simulation?

james-yoo, 2023-08-06:
@yuokamoto 
Thanks for your quick response. The '/clock' topic was showing 60 Hz, and the fps in Unreal Editor was also showing 60. I am currently testing under UE 5.2.1 and figured out there is a framerate issue. I set the 'Fixed Frame Rate' parameter to 100 as below: 
[image]
However, I've noticed that no matter what settings(VSM, Lumen, Nanite, and so on) I change, it always caps at 60 FPS(even in empty black scenes). 
[image]
I will need to check this further, but I believe this is an issue with Unreal Engine itself.

yuokamoto, 2023-08-12:
Thanks for your investigation.
I'm not sure the root cause of the issue. 

FYI
Since UE's frequency is not normally high, e.g. 30hz, I created this custom timer and used inside from publisher.
https://github.com/rapyuta-robotics/rclUE/blob/7613773cd4c1226957603d705d68a2d2b4a69166/Source/rclUE/Public/rclcUtilities.h#L174

james-yoo, 2023-08-14:
@yuokamoto 
Thansk for the response. This issue was mistake on my side and resolved by use UBT on compile time.
I apologies for the inconvenience.

closed 2023-08-14 by james-yoo

pabsan-0, 2024-07-02:
@james-yoo Hi, could you please give more details on this? Do you mean compiling the world and running it in isolation rather than in the editor?