rapyuta-robotics/RapyutaSimulationPlugins #200: Time Sync

Hisham178, 2023-08-22:
How do we sync time, because I noticed time from odom and sensors up untill nano seconds. But from blueprint only in seconds?

yuokamoto, 2023-08-22:
1) Since nano seconds is defined as unsigned int which is not supported in BP, nano seconds can not be exposed.
2) Since time in header is fill from UE time, such as `UGameplayStatics::GetTimeSeconds(GetWorld())`, you can get/save time when odom is published inside UE.

Hisham178, 2023-08-29:
I just found out, you can get up to nano second by doing this. With sim time is true and clock topic. I can finally sync my time across all my nodes

[image]

closed 2023-09-16 by yuokamoto