rapyuta-robotics/RapyutaSimulationPlugins #203: Is possible to use a 3D LiDAR in a RRBaseRobot to detect the movable parts of the robot itself?

dbarrita, 2023-09-05:
Hi, I'm using the plugin to create a few RRBaseRobot and I have put 3D LiDARS on them. The LiDAR works properly the problem if that doesn't detect the elements of the RRBaseRobot itself (car with black wheels in the image). If I want to use LiDAR for detects the position of a movable part of my robot

[image]

How can I put the LiDAR as part of the RRBaseRobot and detect the robot parts? I need to create another RRBaseRobot that only have the LiDAR and attach it to my robot?

Thanks for your time and for develop this plugin.

yuokamoto, 2023-09-05:
I merged one PR. Could you try again?
https://github.com/rapyuta-robotics/RapyutaSimulationPlugins/pull/196

it maybe just visualize lidar point always top

dbarrita, 2023-09-05:
Thanks for the quick response, I have made the following steps:

- Make a pull of the devel branch
- Run the .../GenerateProjectFiles.sh file
- Run $ make (_my_package_name_)Editor

After that the problem persists the sensor detects other RRBaseRobot elements but not the meshes that belongs to the same RRBaseRobot. I check it on RViz and the pointcloud published is the same that it's seen in Unreal

yuokamoto, 2023-09-16:
I created PR. Please check
https://github.com/rapyuta-robotics/RapyutaSimulationPlugins/pull/210

dbarrita, 2023-09-18:
I have tested with LiDAR 3D and 2D and the problem persists. I have tried changing the new variable "_Ignore Self_" in the editor to false but the result is the same as when it's true.

[image]
I have added a little shape to the turtlebot example to test the changes.

dbarrita, 2023-09-18:
Thanks! It works with the 2D LiDAR but I'm trying with the 3D and it doesn't work. Maybe I'm doing something wrong.

https://github.com/rapyuta-robotics/RapyutaSimulationPlugins/assets/90312492/e2c44a9a-b0fe-4850-881d-ba8f3935b378

dbarrita, 2023-10-04:
I have been trying the LiDAR 3D but I can't make it work. With the 2D the behavior is the expected, with the "Ignore Self" parameter I can select the detection of the robot or not. But with the 3D doesn't work. I'm using the actual devel branch, because I have seen that the #210 is merged.

yuokamoto, 2023-10-04:
Thanks for reporting. Let me check later.

yuokamoto, 2023-10-09:
Found bug and created PR
https://github.com/rapyuta-robotics/RapyutaSimulationPlugins/pull/214

dbarrita, 2023-10-09:
Solved! Now it's working properly. Thank you for your time.
[image]

closed 2023-10-09 by dbarrita