rapyuta-robotics/RapyutaSimulationPlugins #205: Build error on UE5.3 branch

james-yoo, 2023-09-11:
@Tadinu 
Thank you for your support on UE 5.3! During compilation using the UE5.3 branch, the following error occurs.

```
/home/username/UnrealProjects/LeoRoverUE5/Plugins/RapyutaSimulationPlugins/Source/RapyutaSimulationPlugins/Public/Core/RRCoreUtils.h:242:13: error: definition of a static variable in a constexpr function is a C++2b extension [-Werror,-Wc++2b-extensions]
/home/sim/UnrealProjects/LeoRoverUE5/Plugins/rclUE/Source/rclUE/Public/logUtilities.h:44:9: note: expanded from macro 'UE_LOG_WITH_INFO'
        UE_LOG(CategoryName, Verbosity, TEXT("[%s] %s"), *__LOG_INFO__, *FString::Printf(__VA_ARGS__)); \
        ^
/home/sim/UnrealEngine_5.3.0/Engine/Source/Runtime/Core/Public/Logging/LogMacros.h:247:3: note: expanded from macro 'UE_LOG'
                UE_PRIVATE_LOG(PREPROCESSOR_NOTHING, constexpr, CategoryName, Verbosity, Format, ##__VA_ARGS__)
                ^
/home/username/UnrealEngine_5.3.0/Engine/Source/Runtime/Core/Public/Logging/LogMacros.h:284:66: note: expanded from macro 'UE_PRIVATE_LOG'
                static constexpr ::UE::Logging::Private::FStaticBasicLogRecord LOG_Static(Format, __builtin_FILE(), __builtin_LINE(), ::ELogVerbosity::Verbosity, LOG_Dynamic); \

/home/sim/UnrealProjects/LeoRoverUE5/Plugins/RapyutaSimulationPlugins/Source/RapyutaSimulationPlugins/Public/Core/RRCoreUtils.h:242:13: error: definition of a static variable in a constexpr function is a C++2b extension [-Werror,-Wc++2b-extensions]
/home/sim/UnrealProjects/LeoRoverUE5/Plugins/rclUE/Source/rclUE/Public/logUtilities.h:44:9: note: expanded from macro 'UE_LOG_WITH_INFO'
        UE_LOG(CategoryName, Verbosity, TEXT("[%s] %s"), *__LOG_INFO__, *FString::Printf(__VA_ARGS__)); \
        ^
/home/sim/UnrealEngine_5.3.0/Engine/Source/Runtime/Core/Public/Logging/LogMacros.h:247:3: note: expanded from macro 'UE_LOG'
                UE_PRIVATE_LOG(PREPROCESSOR_NOTHING, constexpr, CategoryName, Verbosity, Format, ##__VA_ARGS__)
                ^
/home/sim/UnrealEngine_5.3.0/Engine/Source/Runtime/Core/Public/Logging/LogMacros.h:284:66: note: expanded from macro 'UE_PRIVATE_LOG'
                static constexpr ::UE::Logging::Private::FStaticBasicLogRecord LOG_Static(Format, __builtin_FILE(), __builtin_LINE(), ::ELogVerbosity::Verbosity, LOG_Dynamic); \
```
If I remove the constexpr keyword, it compiles fine. Any suggestion?

Tadinu, 2023-09-11:
Hi, pls make sure you have added `CppStandard = CppStandardVersion.Latest;` in all of `.Build.cs` including project & plugins

Tadinu, 2023-09-11:
Hi, pls make sure you use rclUE's 5.2 branch for this & have added `CppStandard = CppStandardVersion.Latest;` in all of `.Build.cs` including project & plugins

james-yoo, 2023-09-11:
@Tadinu 
Thanks for quick response! Issue resolved.

closed 2023-09-11 by james-yoo