rapyuta-robotics/RapyutaSimulationPlugins #209: Is there any way to publish transform?

james-yoo, 2023-09-15:
Hi,
According to the [document](https://rapyutasimulationplugins.readthedocs.io/en/devel/overview.html), ARRBaseRobot has only basic interface such as /cmd_vel, /odom, /joint_states , /joint_commands.  I'd like to publish a transform (such as /tf or /static_tf) for a custom robot, but I can't find a way to do so.

Since Unreal's static meshes correspond to urdf's 'link' attribute, and physics components correspond to urdf's 'joint' attribute, isn't that possible to publish the transform automatically based on the hierarchy of robot actor?

yuokamoto, 2023-09-16:
Hi,
There are multiple options
1.  You can use [robot_state_publisher](http://wiki.ros.org/robot_state_publisher)  which subscribes /joint_states and publish /tf but requires urdf as robot_description
2.  Implement tf publisher using [TFPublisher](https://rapyutasimulationplugins.readthedocs.io/en/devel/doxygen_generated/html/d2/d2a/class_u_r_r_r_o_s2_t_f_publisher.html). it shouldn't be difficult but may need to think about multiple parameter such as offset from static mesh origin to link origin

yuokamoto, 2023-09-18:
wip branch
https://github.com/rapyuta-robotics/RapyutaSimulationPlugins/tree/joint_tf_publisher

yuokamoto, 2023-10-09:
Sorry for taking long.
Created PR https://github.com/rapyuta-robotics/RapyutaSimulationPlugins/pull/213

james-yoo, 2023-10-15:
Hi @yuokamoto,
I would like to know how to apply this 'joint tf publisher' to a mobile robot. A simple screenshot or documentation update would be great.

yuokamoto, 2023-10-15:
Hi,
- need to enable bPublishJointTfPublisher of RRROS2Interface.
https://rapyutasimulationplugins.readthedocs.io/en/devel/doxygen_generated/html/d6/d64/class_u_r_r_robot_r_o_s2_interface.html#af208c8352e4fe16c84fa837afc8e5966
- If joint is made from RRJointComponent, you can publish tf by adding Links and Joints from following methods. Please check BP_PandaArm's construction script as a example. 
https://rapyutasimulationplugins.readthedocs.io/en/devel/doxygen_generated/html/df/d13/class_a_r_r_base_robot.html#a8201f7342a882d457ed9e5d36cb4d14f
https://rapyutasimulationplugins.readthedocs.io/en/devel/doxygen_generated/html/df/d13/class_a_r_r_base_robot.html#aa9671e0d7ab5981d2af32dd58b3937af
- if joint is not RRJointComponent, I added PR to add publish tf between links. https://github.com/rapyuta-robotics/RapyutaSimulationPlugins/pull/217
[image]

james-yoo, 2023-10-15:
@yuokamoto,
Looks great!  Thanks for your quick response!

closed 2023-10-15 by james-yoo