rapyuta-robotics/RapyutaSimulationPlugins #211: Simulation time slows down for low FPS

samiamlabs, 2023-10-04:
Hi!

I would like the physics to run in real-time even for low FPS, but this does not seem to be the case in a project based on the turtlebot3 example that I'm working on. Time slows down for the simulation. I suspect that it has something to do with RRLimitRTFFixedSizeCustomTimeStep?

[image]

Is there some way to deactivate the custom time step stuff (if that is indeed the problem...)? It would be nice to be able to use timestamps from system time on my sensor messages if possible.

yuokamoto, 2023-10-05:
Hello, thanks for your question.

You can disable RRLimitRTFFixedSizeCustomTimeStep by deleting here https://github.com/rapyuta-robotics/turtlebot3-UE/blob/6f118f8a08f5b5251e458b014d9a8a03c35736ab/Config/DefaultEngineBase.ini#L23

samiamlabs, 2023-10-05:
Thank you for your quick response :)

I tried removing the line: `CustomTimeStepClassName=/Script/RapyutaSimulationPlugins.RRLimitRTFFixedSizeCustomTimeStep`

but I still see the Synchronized thing in the FPS view and the simulation is running slower than real-time
[image]

Do I need to do something besides recompiling for the changes to take effect?

yuokamoto, 2023-10-06:
you need to
- restart editor with run_editor.sh since the script generate DefafultEngine.ini
or
- directoly change setting in editor and replay
[image]

samiamlabs, 2023-10-06:
Restarting after removing the line sets the Custom TimeStep to None in Project settings and removes the text in the FPS widget. Thanks!

[image]
[image]

Unfortunately, this did not seem to fix the issue with, for example, `UGameplayStatics::GetTimeSeconds(GetWorld())` not returning seconds in "real-time".
I tried removing everything from the world and that just makes the time run faster than real-time.
If I create a fresh project without any ROS2 stuff, `UGameplayStatics::GetTimeSeconds(GetWorld())` seems to return seconds that run in real-time. 

Any ideas about why this is happening?

yuokamoto, 2023-10-06:
I'm not 100% sure but I think it is depends of UE project framerate setting you can check
https://github.com/rapyuta-robotics/turtlebot3-UE/blob/6f118f8a08f5b5251e458b014d9a8a03c35736ab/Config/DefaultEngineBase.ini#L19
https://docs.unrealengine.com/4.27/en-US/TestingAndOptimization/PerformanceAndProfiling/SmoothFrameRate/

samiamlabs, 2023-10-09:
Oh, sorry. I meat that I set it to False...

yuokamoto, 2023-10-10:
hmm
- if it is slow even  if you use variable time step, I'm not sure but it is more about UE fps setting. it is better to dig into UE more or remove asset from scene and etc
- RRLimitRTFFixedSizeCustomTimeStep support limiting RTF(real time factor) with fixed time step. it may worth tyring to use RRLimitRTFFixedSizeCustomTimeStep and make _TargetRTF_=1 and make _FixedFrameRate_ lower
https://github.com/rapyuta-robotics/turtlebot3-UE/blob/6f118f8a08f5b5251e458b014d9a8a03c35736ab/Config/DefaultEngineBase.ini#L20