rapyuta-robotics/RapyutaSimulationPlugins #225: Support depth camera

james-yoo, 2023-11-13:
Hi rapyuta-robotics team,
Currently, RapyutaSimulationPlugins only supports CameraComponent for RGB image. Any plan to support depth image?

yuokamoto, 2023-11-13:
RapyutaSimulationPlugins has 3d lidar plugin. I'm not sure but can't we use 3d lidar plugin for deth camera as well?

james-yoo, 2023-11-13:
@yuokamoto 
3DLiDARComponent uses 'AsyncLineTraceByChannel' method to check collision between ray and object, it is not suitable for obtaining dense depth image like below:
[image]

yuokamoto, 2023-11-14:
I can create something by changing camera postprocess to SceneDepth. 
it is easy in bp but not sure  how I can set same thing in C++, will investigate more alter

[image]

yuokamoto, 2023-11-16:
https://github.com/rapyuta-robotics/RapyutaSimulationPlugins/pull/231