rapyuta-robotics/RapyutaSimulationPlugins #232: Empty odom data is published

james-yoo, 2023-11-17:
Hi Rapyuta-robotics team,
When I placed [BP_RRTurtlebotWaffle](https://github.com/rapyuta-robotics/RapyutaSimulationPlugins/blob/devel/Content/Robots/Turtlebot3/Physics/BP_RRTurtlebotWaffle.uasset) on level and started simulation, /odom topic is published with empty data as below. Is there anything else I need to set up to receive odom data?
```
header:
  stamp:
    sec: 0
    nanosec: 0
  frame_id: ''
child_frame_id: ''
pose:
  pose:
    position:
      x: 0.0
      y: 0.0
      z: 0.0
    orientation:
      x: 0.0
      y: 0.0
      z: 0.0
      w: 1.0
  covariance: '<array type: double[36]>'
twist:
  twist:
    linear:
      x: 0.0
      y: 0.0
      z: 0.0
    angular:
      x: 0.0
      y: 0.0
      z: 0.0
  covariance: '<array type: double[36]>'
```

james-yoo, 2023-11-18:
@yuokamoto 
OK, I resolved the issue. I have tested under the following conditions.
- Ubuntu 22.04
- Unreal Engine 5.3.2
- RapyutaSimulationPlugins(branch: UE5.3)
- rclUE(branch: UE5_devel_humble)

I started with a new C++ project and cloned the plugins to the project folder. To get it to work, I had to cherry-pick four commits from the 'devel' branch because those commits were merged before the GitHub [action](https://github.com/rapyuta-robotics/RapyutaSimulationPlugins/pull/221) was applied(I think there's something wrong while cherry-picking that didn't output the odom data.).

I created [PR](https://github.com/rapyuta-robotics/RapyutaSimulationPlugins/pull/235) regarding this issue.

yuokamoto, 2023-11-18:
Thank you! I merged PR.

closed 2023-11-18 by yuokamoto