rapyuta-robotics/RapyutaSimulationPlugins #249: What does the variable 'bPublishOdomTf' do?

james-yoo, 2024-01-11:
@yuokamoto 
When I check this option to true in the RRobotROS2Interface, I expected a tf from 'odom' to 'base_footprint' to be published, but nothing happened. What is this variable for?

yuokamoto, 2024-01-12:
yes. that should be. Turtlebot did that as well.
Did you set `bPublishOdom` and `Robot->bMobileRobot` true as well?
https://github.com/rapyuta-robotics/RapyutaSimulationPlugins/blob/d382ad556b6be3d6af1743a9364720616086bbc4/Source/RapyutaSimulationPlugins/Private/Robots/RRRobotROS2Interface.cpp#L43

james-yoo, 2024-01-13:
@yuokamoto 
Thanks for your reply. I tested with 4 robots as below(in Content/Robots/Turtlebot3/Physics) :
- BP_RRTurtlebotBurger
- BP_RRTurtlebotWaffle
- BP_TurtlebotBurger
- BP_TurtlebotWaffle  
  
Although I checked the 'PublishOdomTf' option and started the simulation, there was no transform from 'odom' to 'base_footprint'.  There are only transforms from 'base' to 'wheel_left' and 'wheel_right' as below:
[image]

yuokamoto, 2024-01-13:
Which branch are you using?

This is my case with jammy branch of turtlebot3-UE
[image]

james-yoo, 2024-01-13:
@yuokamoto 
I have cloned 'jammy' branch and checkout following branches:
- RapyutaSimulationPlugins: UE5.3
- rclUE: UE5_devel_humble
  
I am developing with the following conditions:
- Ubuntu 22.04
- Unreal Engine 5.3.2

yuokamoto, 2024-01-14:
Created PR
https://github.com/rapyuta-robotics/RapyutaSimulationPlugins/pull/253

and turtlebot3-UE branch for UE5.3
https://github.com/rapyuta-robotics/turtlebot3-UE/tree/jammy_UE5.3

I still don't know why it works for UE5.1 though

james-yoo, 2024-01-16:
@yuokamoto 
Issue resolved by PR [#253](https://github.com/rapyuta-robotics/RapyutaSimulationPlugins/pull/253). Transform from 'odom' to 'base_footprint' is now publishing.

closed 2024-01-16 by james-yoo