rapyuta-robotics/RapyutaSimulationPlugins #263: Where is the "SceneDepth" material come from?

james-yoo, 2024-02-16:
Hi @yuokamoto,

I would like to control a depth map of the ROS2CameraComponent. It uses the "SceneDepth" material for blending as below:

https://github.com/rapyuta-robotics/RapyutaSimulationPlugins/blob/be7a97715340d189a2a358aece86cfb8714f0384/Source/RapyutaSimulationPlugins/Private/Sensors/RRROS2CameraComponent.cpp#L27

Where is this material come from?

Cheers

yuokamoto, 2024-02-17:
I think it is comes  from here
https://github.com/EpicGames/UnrealEngine/blob/072300df18a94f18077ca20a14224b5d99fee872/Engine/Config/BaseEngine.ini#L2738

[image]

james-yoo, 2024-02-17:
@yuokamoto 
Thank you so much.

closed 2024-02-18 by james-yoo