rapyuta-robotics/RapyutaSimulationPlugins #271: Is it possible to create a GPS using BP?

dbarrita, 2024-03-14:
Hi,
I'm working on using a GPS sensor in my simulation. For this I have georeferenced my  level following the instructions in the unreal [documentation](https://docs.unrealengine.com/5.0/en-US/georeferencing-a-level-in-unreal-engine/). I have obtained the latitude, longitude and altitude of my robot using my robot's blueprint:
[image]
Now I want to publish this data in a ROSNavSatFix message but i have seen that I can't set the latitude, longitude and altitude using the blueprint because in the [source code](https://github.com/rapyuta-robotics/rclUE/blob/devel/Source/rclUE/Public/Msgs/ROS2NavSatFix.h) the option "BlueprintReadWrite" isn't set in the UPROPERTY. Is there a way to publish the NavSatFix using the blueprint or do I have to program it in C++? Thanks in advance

yuokamoto, 2024-03-14:
Hi, 
The main issue of this is BP does not support double
https://docs.unrealengine.com/4.26/en-US/ProgrammingAndScripting/Blueprints/UserGuide/Variables/

Currently you need to program in C++ or publish as a different msg and parse in subscriber side.

yuokamoto, 2024-03-14:
Do you think setter/getter ftom/to float in BP useful for these cases?

dbarrita, 2024-03-14:
Hi, thanks for the quick reply! To answer your question, I don't know. I have experience programming in C++ but not in Unreal, so that is the only reason I used BP. I will try to program it in C++ then.

closed 2024-03-23 by yuokamoto

yuokamoto, 2024-04-06:
FYI:
I created PR
https://github.com/rapyuta-robotics/rclUE/pull/139