rapyuta-robotics/RapyutaSimulationPlugins #276: Create custom sensors components

dbarrita, 2024-04-03:
Hi,
I want to create my own list of sensors and save it in blueprints so I can copy the folder that contains all in each project and don't have to configure the properties and add the meshes in each project/robot. I want this BP to contain the mesh of the sensor and the RR3DLidar component configured with the properties of that sensor. I have seen that in the turtlebot example to add a sensor you have to add the mesh first and later the RR3DLidar component. Is there a way to create one component that contains both? Thanks in advance!

yuokamoto, 2024-04-04:
https://github.com/rapyuta-robotics/RapyutaSimulationPlugins/assets/16463135/e3e358e7-0a4b-4476-aa89-407c0448848b

Hello, you can create Actor which has both mesh and sensor.

yuokamoto, 2024-04-06:
I see. I can't find the option to edit in window...
it is not ideal but I can edit  variable  from construction script
[image]

dbarrita, 2024-04-08:
Hi again,
Finally what I have done is created variables for the color and the topic name in the sensor BP and I have seen that can be edited in the Child Actor Template.
[image]

closed 2024-04-08 by dbarrita