rapyuta-robotics/RapyutaSimulationPlugins #279: Could not find the skeleton for Anim Blueprint

james-yoo, 2024-05-29:
Hi, @yuokamoto,

When I open the 'ABP_RRMany' Animation Blueprint, following error occurs:  
[image]
  
This issue causes the following errors when packaging the game.
```
Error: [AssetLog] /home/user/UnrealProjects/turtlebot3-UE/Plugins/RapyutaSimulationPlugins/Content/Character/ABP_RRManny.uasset: [Compiler] ABP_RRManny  - The skeleton asset for this animation Blueprint is missing, so it cannot be compiled!
```

I have tested under the following condition.

- Ubuntu 22.04
- UE 5.3.2
- turtlebot3-UE (branch: jammy_UE5.3)
- rclUE (branch: UE5_devel_humble)
- RapyutaSimulationPlugins (branch: UE5.3)

yuokamoto, 2024-06-09:
I update UE5.3 and confeirmed packaging work in jammy_UE5.3

james-yoo, 2024-06-19:
Close issue #283

closed 2024-06-19 by james-yoo