rapyuta-robotics/RapyutaSimulationPlugins #282: Support UE5.4

james-yoo, 2024-06-02:
@yuokamoto 
Is there any plan to support UE5.4 soon? If not, I have modified the code to work in UE5.4([link](https://github.com/james-yoo/RapyutaSimulationPlugins/tree/UE5.4)) and it works well as shown in the video below. If you could create a UE5.4 branch, I would like to make PR.  

[image]

I have tested under the following conditions:
- Ubuntu 22.04
- Unreal Engine 5.4.1
- ROS 2 Humble
- rclUE (branch: UE5_devel_humble)

yuokamoto, 2024-06-02:
Thank you!
I created UE5.4 branch
https://github.com/rapyuta-robotics/RapyutaSimulationPlugins/tree/UE5.4

james-yoo, 2024-06-19:
Close issue #284

closed 2024-06-19 by james-yoo