rapyuta-robotics/RapyutaSimulationPlugins #361: How to publish a CameraInfo topic next to a Image

bastianhjaeger, 2025-03-12:
With ROS2 cameras, usually there is a CameraInfo message published with each image. 

E.g. 
```
/camera/image_raw
/camera/camera_info
```

How can I do this with https://github.com/rapyuta-robotics/RapyutaSimulationPlugins ?
I see that rclUE has some UROS2CameraInfoMsg, but not suree how to use.

yuokamoto, 2025-03-13:
Currently camera component does not have publisher for those msg. 
it is required to  add those publisher to CameraComponent t
https://github.com/rapyuta-robotics/RapyutaSimulationPlugins/blob/devel/Source/RapyutaSimulationPlugins/Private/Sensors/RRROS2CameraComponent.cpp

bastianhjaeger, 2025-03-26:
I am willing to implement such a thing. Maybe you can give me some pointers on how I would add a 2nd publishing entity to https://github.com/rapyuta-robotics/RapyutaSimulationPlugins/blob/devel/Source/RapyutaSimulationPlugins/Private/Sensors/RRROS2CameraComponent.cpp

yuokamoto, 2025-03-27:
Thanks for your help.
https://github.com/rapyuta-robotics/RapyutaSimulationPlugins/blob/01c5cc8d26949a20741bebfb7d17d6cd7c5982b2/Source/RapyutaSimulationPlugins/Private/Sensors/RRROS2OverlapSensorComponent.cpp#L25C5-L25C19

This overlapsensor has additional publisher